osresearch/pixel-wrangler #3: EDID emulator

osresearch, 2023-02-18:
The EDID format is awful, but we need to support it anyway...

Adafruit has some example EDID roms that do 800x600 and other formats; maybe we can hack one of those to implement the 640x480@60 timings.

ladyada, 2023-02-18:
i have lost significant life to EDID hacking - just FYI...some OS's 'cache' EDIDs: linux does, windows doesn't seem to. so if you're wondering why it doesn't reload, thats why. there's lots of EDID editors/generateors/extractors tools used in the AV world that are downloadable for free (google for edid editor)

osresearch, 2023-02-18:
Thanks for the advice!

I'm surprised there isn't any online editor, only only decoders.  I can't get the deltacast one to run under wine, so i'll need to spinup a real windows vm to try it out.

osresearch, 2023-02-19:
Hacking in a soft i2c core since I can't figure out Lattice's i2c device and it seems that icestorm devs don't recommend it anyway...

```
[3555446.841541] i915 0000:00:02.0: HDMI-A-2: EDID is invalid:
[3555446.841543]        [00] BAD  00 ff ff ff ff ff ff 00 5a 74 01 00 57 04 00 00
[3555446.841544]        [00] BAD  0a 21 01 04 a2 00 00 78 00 00 00 00 00 00 00 00
[3555446.841544]        [00] BAD  00 00 00 20 00 00 31 40 01 00 01 00 01 00 01 00
[3555446.841545]        [00] BAD  01 00 01 00 01 00 d5 09 80 a0 20 c1 ff ff ff ff
[3555446.841545]        [00] BAD  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[3555446.841545]        [00] BAD  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[3555446.841546]        [00] BAD  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[3555446.841546]        [00] BAD  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[3555451.384838] i2c i2c-1: sendbytes: NAK bailout.
[3555455.772344] i2c i2c-1: sendbytes: NAK bailout.
```

osresearch, 2023-02-19:
for future reference, i2c eeprom address is 0x50.  getting closer!
```
sudo get-edid -i -b 1 | parse-edid
1
This is read-edid version 3.0.2. Prepare for some fun.
Attempting to use i2c interface
Only trying 1 as per your request.
256-byte EDID successfully retrieved from i2c bus 1
Looks like i2c was successful. Have a good day.
WARNING: Checksum failed
Trying to continue...
Section "Monitor"
	Identifier "Pixe���angler"
	ModelName "Pixe���angler"
	VendorName "VST"
	# Monitor Manufactured week 10 of 2023
	# EDID version 1.255
	# Digital Display
	# Display Physical Size not given. Normal for projectors.
	Gamma 2.20
	Option "DPMS" "false"
	#Not giving standard mode: 640x480, 60Hz
	#Not giving standard mode: 256x160, 60Hz
	#Not giving standard mode: 256x160, 60Hz
	#Not giving standard mode: 256x160, 60Hz
	#Not giving standard mode: 256x160, 60Hz
	#Not giving standard mode: 256x160, 60Hz
	#Not giving standard mode: 256x160, 60Hz
	#Not giving standard mode: 256x160, 60Hz
EndSection
```

osresearch, 2023-02-19:
oh no, it looks like to be HDMI compliant the i2c lines are supposed to be pulled up to 5V, not 3.3V.  the ice40up5k isn't 5v tolerant, so that would require a bidirectional level shifter on the SDA line (the SCL is input only).  this would complicate things.

osresearch, 2023-02-20:
Hurrah! the i2c device now properly implements the repeated START state and delivers an error free EDID, although x isn't swithcing to it for some reason.

https://github.com/osresearch/pixel-wrangler/commit/0d43c9e86c644c38e9c9a46d856a12ce9686b799

osresearch, 2023-02-20:
Linux (x86) can be forced to use the HDMI even though it doesn't like the EDID:

```
echo on-digital | sudo tee /sys/class/drm/card0-HDMI-A-2/status 
xrandr --output HDMI-2 --mode 640x480
```

Hot plug isn't doing the right thing, although if you mess with the pullup resistor it can cause the kernel to decide that there is a drm interrupt storm.

The adafruit 800x480 EDID almost works, although we can't handled the 32 MHz pixel clock yet.

osresearch, 2023-02-20:
I think the EDID emulation works well enough now:
```
sudo get-edid  -b 1 | parse-edid
1
This is read-edid version 3.0.2. Prepare for some fun.
Attempting to use i2c interface
Only trying 1 as per your request.
256-byte EDID successfully retrieved from i2c bus 1
Looks like i2c was successful. Have a good day.
Checksum Correct

Section "Monitor"
	Identifier "PixelWrangler"
	ModelName "PixelWrangler"
	VendorName "PXL"
	# Monitor Model Year: 2023
	# EDID version 1.4
	# Digital Display
	DisplaySize 640 480
	Gamma 2.20
	Option "DPMS" "false"
	Modeline 	"Mode 0" 25.17 640 656 752 800 480 490 492 525 -hsync -vsync 
EndSection
```

The problem appears to be with hotplug or with X rejecting the 640x480 display parameters?  Let's move that to a new issue.

closed 2023-02-20 by osresearch