osresearch/pixel-wrangler #22: v0 board issues

osresearch, 2023-02-26:
* [ ] L1 is too close to cap on the top
* [ ] Reset and SW1 should be brought to the headers
* [ ] Two IOs are not routed out
* [ ] USB footprint isn't quite right
* [ ] USB resistors are on the top; move to bottom?
* [ ] Add a clock?
* [ ] Reduce size of paste window under ice40
* [ ] Make sure to order ENIG plating!
* [ ] 1.2mm thick PCB
* [ ] Smaller stencil size

cibomahto, 2023-02-26:
My suggestions:
* Change button type to 'TS-1185-C-A-A': https://www.lcsc.com/product-detail/Others_XKB-TS-1185-C-A-A_C318896.html (slightly larger but more robust/very common)
* Change USB type to 'MICRO XNJ ZB' https://www.lcsc.com/product-detail/USB-Connectors_SHOU-HAN-MICRO-XNJ-ZB_C668587.html (this works with the current footprint?)
* Suggest 2520 oscillator package size, for example: https://www.lcsc.com/product-detail/Oscillators_Yangxing-Tech-OT252012MJBA4SL_C669073.html
* Leave USB resistors on top side (keep USB lines on a single plane to avoid discontinuities)
* Consider if the other end of the I2C pins are being pulled up to 5V, which would make a voltage conflict.

osresearch, 2023-02-26:
Thanks for the suggestions!

Software patches are being developed in https://github.com/osresearch/pixel-wrangler/tree/v0-fixes

osresearch, 2023-02-27:
For the i2c pullups, perhaps using a device-side version of the [TPD12S016 "HDMI Companion chip"](https://nl.mouser.com/ProductDetail/Texas-Instruments/TPD12S016RKTR) would be a good idea.  It adds ESD protection on all eight differential pairs, i2c level shifting, etc.

osresearch, 2023-02-27:
After soldering a jumper from the SDA pullup resistor to GPIO 0 7, the i2c works and the laptop negotiates with the display.  All three channel decode works.  Green is inverted on schematic, but doesn't get inverted in the verilog?  Something seems wrong with that.

osresearch, 2023-12-07:
[image]

I forgot to check all the replies to this issue and forgot to change the switch or USB footprints.  I did fix the i2c connections and moved everything to one side for easier manufacturing.

Aisler is doing the prototype and they only have thick PCBs.

osresearch, 2024-02-10:
And some v2 board issues:

The biggest:

* The v2 boards won't sync up their PLL, so they never get the video (maybe the througholes are a bad idea)
* Maybe this is a bad idea and it should just use an HDMI chip

And some other ones:

* Not having an oscillator means that free running video outputs have some shimmer (on v0 and v2).
* Buttons are still too hard to press
* USB micro rips off sometimes
* Initial program load is a problem; have to figure out how to make it work on populated boards
* Boot loader doesn't time out and start video automatically
* Boot loader metadata isn't populated correctly