osresearch/pixel-wrangler #29: Availability of the pixel-wrangler hardware

stitch, 2023-03-12:
Hi Trammell, 

I'm super impressed with this project. I'd like to use it as an easy way to drive a pixel display in my cyber coin pusher at the end of the year ccc conference. Do you plan this hardware to be going into production sometime?

https://twitter.com/TheCyberPusher

Regards,
Elger / Stitch

osresearch, 2023-03-12:
Production plans are uncertain, although I have an unpopulated v0 board if you want to hack on it.  It works, although there are some issues remaining with the HDMI decode. Some place-and-routes are totally stable and produce beautiful glitch free framebuffer imgaes, while others are nothing but static.  I've been chasing this for the past two weeks -- we're pushing the limits of the ice40 fabric -- and all of the demo videos are from the lucky ones that worked well.

So if you have some tolerance for instability during testing and finding a bitstream that gives you a solid decoding, you're welcome to one of the boards!

osresearch, 2023-03-23:
[image]

The glitch branch is looking much better, so I'm optimistic that we have the timing issues sorted.  Any chance you'll be at the hackaday berlin meetup this weekend? I have a self-contained demo board to show off.